radxa/kernel #37: btrfs support on stable-5.10-rock5

jpalus, 2022-10-30:
Would it be possible to enable `btrfs` support for rock5?

RadxaYuntian, 2022-10-31:
5B will be supported under `linux-5.10-gen-rkr3.4` branch in the future. In this release Rockchip also added supports for various their SoCs so we can have a unified kernel for most of our Rockchip products.

With this change we are also going to switch to our new build systems. I have added [btrfs support](https://github.com/radxa-repo/bsp/commit/a79155240f3d75fd165a24912ead86db6ebcb41a#diff-bca425fb8832cacf9174ea59fb1812586ac03b1299c811d8f7af8b6dbb49c92cR286-R288) in our `bsp` build tool.

closed 2022-10-31 by RadxaYuntian

hombit, 2022-12-01:
May I ask when the kernel with BTRFS support for 5B comes to the apt repo?

RadxaYuntian, 2022-12-02:
Currently the schedule is to testing ROCK 4 images for the next 2 weeks, then ROCK 5 images for another 2 weeks. After which hopefully we can release this new image along with the updated apt repo.

marvin-enthus, 2023-04-13:
May I ask about the status of it? I Just had a look on https://github.com/radxa/kernel/blob/linux-5.10-gen-rkr3.4/arch/arm64/configs/rockchip_linux_defconfig - BTRFS still seems to be disabled?

RadxaYuntian, 2023-04-14:
Our prebuilt kernel will have it enabled: https://github.com/radxa-repo/bsp/blob/main/linux/.common/kconfig.conf#L315-L317