radxa/kernel #44: Please disable the stupid gcc wrapper

adam900710, 2022-11-23:
The wrapper introduced in commit c2482146d144 (rk: Makefile: add gcc-wrapper) is a poor man's -Werror build config.

While the RK guys choose not to backport the proper config for it, they introduced their own wrapper for it.

The damn wrapper can not even handle the latest toolchain, as all the newer warnings would just be treated as error and fail the build.

If RK guys are not able to do proper backports, please don't be creative on causing new bugs.

adam900710, 2022-11-23:
OK, Rockchip sh*t never changes.

For BSP, does it support aarch64 native build? Aka, build the whole thing under aarch64 environment (with distcc for help though)

RadxaYuntian, 2022-11-23:
Currently all our developments are done on x86_64 system so this is not tested. You can try building with `-n` flag which disables Docker.

adam900710, 2022-11-23:
All right.

I can already build the vendor kernel after reverting that wrapper commit natively on aarch64 with ArchlinuxARM already.
I guess it should be possible to use `dsp` tool to build the kernel at least.