radxa/kernel #54: ZFS module unable to build

stephenjamieson, 2023-01-04:
Hello, I am attempting to use ZFS on the rock 5b, however I am running into multiple issues that I believe need to be addressed.

The first issue is that some of the scripts are trying to use `/bin/env python` when it should be using `python3` as the python symlink is disabled by default.  Can be fixed by installing `python-is-python3`. Of course if we're expecting to use python2 here, then it should be using `/bin/env python2`

My next roadblock was an error from the DKMS install

```
sudo apt-get install zfs-dkms
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  debhelper
Recommended packages:
  zfs-zed linux-libc-dev
The following NEW packages will be installed:
  zfs-dkms
0 upgraded, 1 newly installed, 0 to remove and 2 not upgraded.
Need to get 0 B/2,346 kB of archives.
After this operation, 18.0 MB of additional disk space will be used.
Preconfiguring packages ...
Selecting previously unselected package zfs-dkms.
(Reading database ... 62832 files and directories currently installed.)
Preparing to unpack .../zfs-dkms_2.1.4-0ubuntu0.1_all.deb ...
Unpacking zfs-dkms (2.1.4-0ubuntu0.1) ...
Setting up zfs-dkms (2.1.4-0ubuntu0.1) ...
Loading new zfs-2.1.4 DKMS files...
Building for 5.10.110-rockchip-rk3588
Building initial module for 5.10.110-rockchip-rk3588
configure: error:
        *** Unable to build an empty module.

Error! Bad return status for module build on kernel: 5.10.110-rockchip-rk3588 (aarch64)
Consult /var/lib/dkms/zfs/2.1.4/build/make.log for more information.
dpkg: error processing package zfs-dkms (--configure):
 installed zfs-dkms package post-installation script subprocess returned error exit status 10
Processing triggers for initramfs-tools (0.140ubuntu13.1) ...
Errors were encountered while processing:
 zfs-dkms

E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Reviewing the config.log:

```
configure:19387:
            KBUILD_MODPOST_NOFINAL= KBUILD_MODPOST_WARN=
            make modules -k -j8    -C /lib/modules/5.10.110-rockchip-rk3588/build
            M=/var/lib/dkms/zfs/2.1.4/build/build/conftest >build/conftest/build.log 2>&1
configure:19390: $? = 2
configure:19393: test -f build/conftest/conftest.ko
configure:19396: $? = 1
configure:19405: result: no
configure:19408: error:
        *** Unable to build an empty module.
```

Attempting to build this manually:

```
KBUILD_MODPOST_NOFINAL= KBUILD_MODPOST_WARN=
            make modules -k -j8    -C /lib/modules/5.10.110-rockchip-rk3588/build
            M=/var/lib/dkms/zfs/2.1.4/build/build/conftest >build/conftest/build.log 2>&1
make: Entering directory '/usr/src/linux-headers-5.10.110-rockchip-rk3588'
  UPD     include/config/kernel.release
  UPD     include/generated/utsrelease.h
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
  CC      scripts/mod/devicetable-offsets.s
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTCC  scripts/mod/file2alias.o
  HOSTLD  scripts/mod/modpost
scripts/Makefile.build:423: warning: overriding recipe for target 'modules.order'
Makefile:1518: warning: ignoring old recipe for target 'modules.order'
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/vdso.lds', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/vgettimeofday.o', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/note.o', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/sigreturn.o', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.
make[1]: Target 'include/generated/vdso-offsets.h' not remade because of errors.
make: *** [arch/arm64/Makefile:194: vdso_prepare] Error 2
make: Target 'modules' not remade because of errors.
make: Leaving directory '/usr/src/linux-headers-5.10.110-rockchip-rk3588'
zsh: no such file or directory: build/conftest/build.log
```

Commenting out the following in `linux-headers-5.10.110-rockchip-rk3588/arch/arm64/Makefile` gets around that error:

```
ifeq ($(KBUILD_EXTMOD),)
# We need to generate vdso-offsets.h before compiling certain files in kernel/.
# In order to do that, we should use the archprepare target, but we can't since
# asm-offsets.h is included in some files used to generate vdso-offsets.h, and
# asm-offsets.h is built in prepare0, for which archprepare is a dependency.
# Therefore we need to generate the header after prepare0 has been made, hence
# this hack.
prepare: vdso_prepare
vdso_prepare: prepare0
       $(Q)$(MAKE) $(build)=arch/arm64/kernel/vdso include/generated/vdso-offsets.h
       $(if $(CONFIG_COMPAT_VDSO),$(Q)$(MAKE) \
               $(build)=arch/arm64/kernel/vdso32  \
               include/generated/vdso32-offsets.h)
endif
```

But commenting out that code leads to:

```
checking whether fops->fallocate() exists... configure: error:
        *** None of the expected "file_fallocate" interfaces were detected.
        *** This may be because your kernel version is newer than what is
        *** supported, or you are using a patched custom kernel with
        *** incompatible modifications.
        ***
        *** ZFS Version: zfs-2.1.4-0ubuntu0.1
        *** Compatible Kernels: 3.10 - 5.17
```

So overall... there's no way to use ZFS on this kernel and likely many other DKMS installed kernel modules.

RadxaYuntian, 2023-01-05:
Can you retest with [this image](https://github.com/radxa-build/rock-5b/releases/download/20230103-0406/rock-5b_debian_bullseye_kde_2023-01-03T0407+0000_gpt.img.xz)?

stephenjamieson, 2023-01-05:
Hi @RadxaYuntian, I was trying the cli image and the default passwords in the FAQ are not working.

RadxaYuntian, 2023-01-06:
We do not test or recommend Ubuntu image, as Rockchip based devices requires Rockchip custom packages to have better hardware support, which is currently available for Debian. However, booting issue was reported to us before and we look to fix it soon to support ROS users (which support Ubuntu only).

I'll check PDE_DATA related kernel config. However, Rockchip kernel is a backported monster splitting off from mainline Linux back in 2.x days, so it might not work after all.

stephenjamieson, 2023-01-30:
Hello, was there any progress on this @RadxaYuntian

RadxaYuntian, 2023-01-30:
I looked at it briefly before and it seems to be missing quite a few parts. It is not trivial and it will take a while before I'm back working on ROCK 5.

linnaea, 2023-04-09:
This is caused by the upstream kernel being derived from Android GKI kernel, which has all VFS symbol exported to a different namespace than where the vanilla kernel exports them to. To build ZFS on this kernel you need to patch zfs.

Open `config/kernel.m4`, search for `MODULE_LICENSE($3);` and add `MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);` below it, then edit `include/os/linux/kernel/linux/mod_compat.h`, look for `ZFS_MODULE_DESCRIPTION`, add `;MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver)` to the end of that line, run `autogen.sh` and it should work.

shivabohemian, 2023-11-01:
> This is caused by the upstream kernel being derived from Android ACK kernel, which has all VFS symbol exported to a different namespace than where the vanilla kernel exports them to. To build ZFS on this kernel you need to patch zfs.
> 
> Open `config/kernel.m4`, search for `MODULE_LICENSE($3);` and add `MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);` below it, then edit `include/os/linux/kernel/linux/mod_compat.h`, look for `ZFS_MODULE_DESCRIPTION`, add `;MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver)` to the end of that line, run `autogen.sh` and it should work.

Hello, I found that `ZFS_MODULE_DESCRIPTION` has been removed in newly released ZFS 2.2 (`include/OS/Linux/kernel/Linux/mod_compat.h`). How should I edit it? Thank you!

linnaea, 2023-11-01:
> Hello, I found that `ZFS_MODULE_DESCRIPTION` has been removed in newly released ZFS 2.2 (`include/OS/Linux/kernel/Linux/mod_compat.h`). How should I edit it? Thank you!

```patch
diff -durN zfs-2.2.0.orig/config/kernel.m4 zfs-2.2.0/config/kernel.m4
--- zfs-2.2.0.orig/config/kernel.m4	2023-10-13 07:03:31.000000000 +0800
+++ zfs-2.2.0/config/kernel.m4	2023-10-21 17:15:21.214263487 +0800
@@ -660,6 +660,7 @@
 MODULE_AUTHOR(ZFS_META_AUTHOR);
 MODULE_VERSION(ZFS_META_VERSION "-" ZFS_META_RELEASE);
 MODULE_LICENSE($3);
+MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
 ])
 
 dnl #
diff -durN zfs-2.2.0.orig/module/os/linux/spl/spl-generic.c zfs-2.2.0/module/os/linux/spl/spl-generic.c
--- zfs-2.2.0.orig/module/os/linux/spl/spl-generic.c	2023-10-13 07:41:34.965111309 +0800
+++ zfs-2.2.0/module/os/linux/spl/spl-generic.c	2023-10-21 17:16:25.231694524 +0800
@@ -929,3 +929,4 @@
 MODULE_AUTHOR(ZFS_META_AUTHOR);
 MODULE_LICENSE("GPL");
 MODULE_VERSION(ZFS_META_VERSION "-" ZFS_META_RELEASE);
+MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
diff -durN zfs-2.2.0.orig/module/os/linux/zfs/zfs_ioctl_os.c zfs-2.2.0/module/os/linux/zfs/zfs_ioctl_os.c
--- zfs-2.2.0.orig/module/os/linux/zfs/zfs_ioctl_os.c	2023-10-13 07:41:34.894111142 +0800
+++ zfs-2.2.0/module/os/linux/zfs/zfs_ioctl_os.c	2023-10-21 17:17:42.042612036 +0800
@@ -377,3 +377,4 @@
 MODULE_LICENSE("Dual BSD/GPL"); /* zstd / misc */
 MODULE_LICENSE(ZFS_META_LICENSE);
 MODULE_VERSION(ZFS_META_VERSION "-" ZFS_META_RELEASE);
+MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
```

shivabohemian, 2023-11-01:
> > Hello, I found that `ZFS_MODULE_DESCRIPTION` has been removed in newly released ZFS 2.2 (`include/OS/Linux/kernel/Linux/mod_compat.h`). How should I edit it? Thank you!
> 
> ```diff
> diff -durN zfs-2.2.0.orig/config/kernel.m4 zfs-2.2.0/config/kernel.m4
> --- zfs-2.2.0.orig/config/kernel.m4	2023-10-13 07:03:31.000000000 +0800
> +++ zfs-2.2.0/config/kernel.m4	2023-10-21 17:15:21.214263487 +0800
> @@ -660,6 +660,7 @@
>  MODULE_AUTHOR(ZFS_META_AUTHOR);
>  MODULE_VERSION(ZFS_META_VERSION "-" ZFS_META_RELEASE);
>  MODULE_LICENSE($3);
> +MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
>  ])
>  
>  dnl #
> diff -durN zfs-2.2.0.orig/module/os/linux/spl/spl-generic.c zfs-2.2.0/module/os/linux/spl/spl-generic.c
> --- zfs-2.2.0.orig/module/os/linux/spl/spl-generic.c	2023-10-13 07:41:34.965111309 +0800
> +++ zfs-2.2.0/module/os/linux/spl/spl-generic.c	2023-10-21 17:16:25.231694524 +0800
> @@ -929,3 +929,4 @@
>  MODULE_AUTHOR(ZFS_META_AUTHOR);
>  MODULE_LICENSE("GPL");
>  MODULE_VERSION(ZFS_META_VERSION "-" ZFS_META_RELEASE);
> +MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
> diff -durN zfs-2.2.0.orig/module/os/linux/zfs/zfs_ioctl_os.c zfs-2.2.0/module/os/linux/zfs/zfs_ioctl_os.c
> --- zfs-2.2.0.orig/module/os/linux/zfs/zfs_ioctl_os.c	2023-10-13 07:41:34.894111142 +0800
> +++ zfs-2.2.0/module/os/linux/zfs/zfs_ioctl_os.c	2023-10-21 17:17:42.042612036 +0800
> @@ -377,3 +377,4 @@
>  MODULE_LICENSE("Dual BSD/GPL"); /* zstd / misc */
>  MODULE_LICENSE(ZFS_META_LICENSE);
>  MODULE_VERSION(ZFS_META_VERSION "-" ZFS_META_RELEASE);
> +MODULE_IMPORT_NS(VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver);
> ```
Thank you very much! You're great.