radxa/kernel #88: Rock 5*A*  fan pwm

Rippanda12, 2023-04-08:
Hello on rock 5B fan pwm is `/sys/devices/platform/fd8b0010.pwm` 
looking using `ls /sys/devices/platform/*pwm*` results in nothing

CodeChenL, 2023-04-11:
try`/sys/class/pwm`

RadxaYuntian, 2023-04-11:
Fan support has been added in the following commit: https://github.com/radxa/kernel/commit/d3e9cd5cda3ac169c663682f29af1e688519798c.

closed 2023-04-11 by RadxaYuntian

Rippanda12, 2023-04-14:
Doesn't work the fan doesnt spin up (the chip almost throthles 70C)
[image]
[image]
meanwhile on rock 5B theres a pwm* folder which allows the controll
[image]

RadxaYuntian, 2023-04-14:
Once the PWM signal is registered as `pwm-fan` it cannot be accessed by standard PWM interface since it is now part of thermal subsystem. You should [set thermal governors](https://github.com/radxa-pkg/rsetup/blob/main/src/usr/lib/rsetup/cli/system.sh#L85-L89) to [`step_wise`](https://github.com/radxa-pkg/rsetup/blob/main/src/usr/lib/rsetup/tui/hardware/hardware.sh#L97) so the fan can be controlled according to the [temperature targets](https://github.com/radxa/kernel/commit/d3e9cd5cda3ac169c663682f29af1e688519798c#diff-ca6356fe35b0685a8e7d552cc799f6cbbf3b5211113571edbb6128f749a67372R702-R722).

Rippanda12, 2023-04-14:
then why isnt the same done for rock 5b?

RadxaYuntian, 2023-04-14:
This is part of the requirement for our new systems, that all PWM fan has to be properly registered as thermal cooling devices. 5B's system was created before this change, so it will be updated later. We expect to release new 5B image along with 5A.