radxa/kernel #121: RTL8156B 2.5G USB Network Card Support Exception

jxhexi, 2023-05-14:
[image]
[image]
[image]
[image]

RadxaStephen, 2023-05-14:
Do you test it on ROCK 5B?
And which image do you use? Please shoe me the download link.
And please provide the full boot log from console tool.

jxhexi, 2023-05-17:
img：rock-5b-ubuntu-focal-server-arm64-20230201-1200-gpt.img.xz
[dmesg.txt](https://github.com/radxa/kernel/files/11500006/dmesg.txt)