radxa/kernel #134: Potential typo in rk3568 dts

Simi4, 2023-06-09:
I found a strange place in the dts files for rk3568 boards:
pinctrl-2 = <&**soc_slppin_gpio**>, <&rk817_slppin_pwrdn>;
pinctrl-3 = <&**soc_slppin_gpio**>, <&rk817_slppin_rst>;

Perhaps pinctrl-3 should refer to this pinconfig: soc_slppin_rst

RadxaYuntian, 2023-06-12:
Can you link the file? BTW duplicated definition is not necessarily a bug, as seen in [this overlay](https://github.com/radxa/overlays/blob/main/arch/arm64/boot/dts/rockchip/overlays/rk3568-spi3-m1-cs0-mcp2515.dts#L50-L52).

Simi4, 2023-06-14:
Here, for example:
https://github.com/radxa/kernel/blob/fdb12912e32f4d615457b3d0cd31d7d201737afd/arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dts#L368-L369