radxa/kernel #140: Trying to build a 5.10 RT kernel

pissalou, 2023-06-19:
I am using the radxa-repo/bsp project to build a 5.10 kernel with the rt patch.
I have configured the project to fetch the kernel sources from https://github.com/radxa/kernel@linux-5.10-gen-rkr3.4 and edited kconfig.conf to have the following parameters set:
```
CONFIG_PREEMPT_NONE=n
CONFIG_PREEMPT=y
CONFIG_PREEMPT_RT=y
CONFIG_PREEMPT_RT_FULL=y
```

The kernel builds fine but after installing it and looking at config file under boot, it still says:
```
radxa@rock-4se:~$ grep PREEM /boot/config-5.10.110-555-rockchip
CONFIG_PREEMPT_NONE=y
# CONFIG_PREEMPT_VOLUNTARY is not set
# CONFIG_PREEMPT is not set
CONFIG_PREEMPT_NOTIFIERS=y
# CONFIG_PREEMPTIRQ_DELAY_TEST is not set
```

full log can be found here: https://github.com/pissalou/bsp/actions/runs/5310459566/jobs/9612468683

closed 2023-06-19 by RadxaYuntian

pissalou, 2023-06-19:
So it means there is no possibility of a fully automated custom kernel build using github actions?
The beauty of the radxa-repo/bsp project is that one can build a kernel without needing an x86 machine or a compiler.

RadxaYuntian, 2023-06-19:
`bsp` is fully capable to build custom kernels, and we use it to build our own kernels that shipped in our system. The issue you have is that you do not have the correct kernel config. For example, to undefine a config, you need to use `# CONFIG_PREEMPT_NONE is not set` syntax. Also for `CONFIG_PREEMPT_RT` you need to define [`[CONFIG_EXPERT](https://cateee.net/lkddb/web-lkddb/EXPERT.html)` and `[CONFIG_ARCH_SUPPORTS_RT](https://cateee.net/lkddb/web-lkddb/ARCH_SUPPORTS_RT.html)`](https://cateee.net/lkddb/web-lkddb/PREEMPT_RT.html). Once you correct your kernel profile you can build like any other kernel we build.

pissalou, 2023-06-19:
Thank you for the clarification, @RadxaYuntian. It is not like I need an interactive menu to be able to create a valid kernel config, it just makes it easier for a noob like me to write it correctly.