radxa/kernel #151: There is a problem with dw-hdmi-qp-cec probe in rkr4.1

unifreq, 2023-07-09:
Model: rock5b
There is a problem with the kernel compiled with rkr4.1 source code:

[   10.343117] ------------[ cut here ]------------
[   10.343129] WARNING: CPU: 5 PID: 439 at kernel/irq/manage.c:2071 request_threaded_irq+0x150/0x190
[   10.343133] Modules linked in: dw_hdmi_qp_cec(+) bifrost_kbase memory_group_manager rknpu fuse ip_tables ipv6 r8125
[   10.343147] CPU: 5 PID: 439 Comm: systemd-udevd Tainted: G        W         5.10.160-rk3588-flippy-2307b #106
[   10.343151] Hardware name: Radxa ROCK 5B (DT)
[   10.343154] pstate: a0400009 (NzCv daif +PAN -UAO -TCO BTYPE=--)
[   10.343158] pc : request_threaded_irq+0x150/0x190
[   10.343162] lr : request_threaded_irq+0x74/0x190
[   10.343165] sp : ffffffc00acc3820
[   10.343168] x29: ffffffc00acc3820 x28: ffffffc00acc3d90 
[   10.343171] x27: ffffffc00117d068 x26: ffffffc00117c310 
[   10.343175] x25: ffffffc00117c310 x24: 0000000000000001 
[   10.343179] x23: ffffff8103a70000 x22: ffffffc00117c1a0 
[   10.343183] x21: ffffff83fc525200 x20: 0000000000000080 
[   10.343186] x19: ffffff8102ecac80 x18: ffffffffffffffff 
[   10.343191] x17: 0000000000000000 x16: 0000000000000000 
[   10.343194] x15: ffffffffffffffff x14: ffffff81036b8307 
[   10.343198] x13: ffffff81036b8305 x12: 00000000fffd0000 
[   10.343202] x11: 0000000000000040 x10: ffffffc009a460e8 
[   10.343206] x9 : ffffffc009a460e0 x8 : ffffff8100000270 
[   10.343209] x7 : 0000000000000000 x6 : 0000000000000000 
[   10.343213] x5 : ffffff8100000248 x4 : ffffff8100000278 
[   10.343216] x3 : 0000000000000000 x2 : 0000000000000000 
[   10.343220] x1 : 0000000000000001 x0 : 0000000000331600 
[   10.343224] Call trace:
[   10.343227]  request_threaded_irq+0x150/0x190
[   10.343231]  devm_request_threaded_irq+0x80/0x100
[   10.343239]  dw_hdmi_qp_cec_probe+0x12c/0x25c [dw_hdmi_qp_cec]
[   10.343244]  platform_drv_probe+0x54/0xac
[   10.343247]  really_probe+0x108/0x520
[   10.343251]  driver_probe_device+0x70/0x160
[   10.343254]  device_driver_attach+0xc0/0xd0
[   10.343258]  __driver_attach+0x108/0x184
[   10.343262]  bus_for_each_dev+0x70/0xd0
[   10.343265]  driver_attach+0x24/0x30
[   10.343268]  bus_add_driver+0x120/0x23c
[   10.343272]  driver_register+0x78/0x130
[   10.343275]  __platform_driver_register+0x48/0x54
[   10.343279]  dw_hdmi_qp_cec_driver_init+0x20/0xff4 [dw_hdmi_qp_cec]
[   10.343283]  do_one_initcall+0x50/0x1c0
[   10.343287]  do_init_module+0x44/0x240
[   10.343289]  load_module+0x2068/0x2810
[   10.343293]  __do_sys_finit_module+0xbc/0x120
[   10.343296]  __arm64_sys_finit_module+0x20/0x30
[   10.343300]  el0_svc_common.constprop.0+0x9c/0x2c0
[   10.343304]  do_el0_svc+0x6c/0x90
[   10.343308]  el0_svc+0x10/0x20
[   10.343310]  el0_sync_handler+0xe8/0x114
[   10.343314]  el0_sync+0x1a0/0x1c0
[   10.343316] 
               PC: 0xffffffc0080f80e0:
[   10.343319] 7ee0  17fffed6 f9402660 f9408c00 362ffe40 17ffff1f aa1303e0 52800002 52800021
[   10.343328] 7f00  94000c1d f9003e74 17ffff49 aa1703e0 d63f0020 17ffff51 a94153f3 12800256
[   10.343336] 7f20  17fffedb a94153f3 128004b6 17fffed8 d4210000 17ffff3b 128002b6 17fffed4
[   10.343343] 7f40  12800176 17fffecf f9402663 2a1503e2 f9402a81 b000a820 f9400463 91086000
[   10.343351] 7f60  128002b6 943dd1a6 17fffe98 2a1503e1 b000aea0 9110e000 943dd1a1 f9000ed4
[   10.343359] 7f80  aa1403e1 aa1303e0 94002722 17ffffb2 d503233f a9ba7bfd 910003fd a90153f3
[   10.343366] 7fa0  a9025bf5 aa0103f6 52b00001 6b01001f 54000b40 a90363f7 f279007f aa0303f4
[   10.343374] 7fc0  aa0503f7 fa4018a0 54000920 92792c61 926ed421 f141003f 540008a0 92721061
[   10.343382] 7fe0  926ef021 f141103f 54000820 2a0003f8 a9046bf9 aa0203fa f9002bfb aa0403fb
[   10.343390] 8000  97fff148 aa0003f5 b4000780 b9408000 37580740 37880660 b40003f6 d000b4c0
[   10.343398] 8020  5281b801 f9424800 9404fac7 aa0003f3 b40006a0 a9005e76 9100c2b9 aa1903e0
[   10.343406] 8040  f900127a b9003e74 f9002a7b 94000fa1 2a0003f4 37f80300 aa1503e1 2a1803e0
[   10.343413] 8060  aa1303e2 97fffe03 2a0003f4 350001c0 a94363f7 a9446bf9 f9402bfb 2a1403e0
[   10.343421] 8080  a94153f3 a9425bf5 a8c67bfd d50323bf d65f03c0 b400033a d503201f 10fecd36
[   10.343429] 80a0  17ffffdf aa1903e0 94000fae f9401a60 9404f3f0 aa1303e0 9404f3ee a9425bf5
[   10.343437] 80c0  2a1403e0 a94153f3 a94363f7 a9446bf9 f9402bfb a8c67bfd d50323bf d65f03c0
[... 608 lines cut ...]
[   10.347788] 3760  ffffffff ffffffff 02ecac80 ffffff81 00000080 00000000 fc525200 ffffff83
[   10.347795] 3780  0117c1a0 ffffffc0 03a70000 ffffff81 00000001 00000000 0117c310 ffffffc0
[   10.347803] 37a0  0117c310 ffffffc0 0117d068 ffffffc0 0acc3d90 ffffffc0 0acc3820 ffffffc0
[   10.347811] 37c0  080f8004 ffffffc0 0acc3820 ffffffc0 080f80e0 ffffffc0 a0400009 00000000
[   10.347818] 37e0  0acc3850 ffffffc0 089c7ad0 ffffffc0 ffffffff 0000007f 00000010 00000000
[   10.347826] 3800  0acc3820 ffffffc0 080f80e0 ffffffc0 00000000 00000000 080f8004 ffffffc0
[   10.347833] 3820* 0acc3880 ffffffc0 080fc2c0 ffffffc0 02ecac80 ffffff81 0117d068 ffffffc0
[   10.347841] 3840  03a70000 ffffff81 00000001 00000000 00d75c10 ffffff81 0117c1a0 ffffffc0
[   10.347848] 3860  0117c310 ffffffc0 00000080 00000000 09ad5370 ffffffc0 00000001 00000000
[   10.347856] 3880  0acc38d0 ffffffc0 0117c4d0 ffffffc0 03c90080 ffffff81 00d75c10 ffffff81
[   10.347863] 38a0  00d75c00 ffffff81 00000000 00000000 0117c2f0 ffffffc0 09b67760 ffffffc0
[   10.347870] 38c0  0000000b 00000000 0117e150 ffffffc0 0acc3910 ffffffc0 089c5748 ffffffc0
[   10.347878] 38e0  00000000 00000000 00d75c10 ffffff81 0117e028 ffffffc0 00000000 00000000
[   10.347885] 3900  0117e028 ffffffc0 15a9aa00 3341f1d4 0acc3940 ffffffc0 089c2a98 ffffffc0
[   10.347893] 3920  00d75c10 ffffff81 09c30180 ffffffc0 00000000 00000000 09c30180 ffffffc0
[   10.347901] 3940  0acc39a0 ffffffc0 089c3310 ffffffc0 00d75c10 ffffff81 09b67760 ffffffc0
[   10.347909] 3960  0117e028 ffffffc0 00000000 00000000 09ad5240 ffffffc0 044cb068 ffffff81
[   10.347916] 3980  0117e100 ffffffc0 0117e150 ffffffc0 00000100 00000000 00000000 00000000
[   10.347923] 39a0  0acc39d0 ffffffc0 089c37d0 ffffffc0 00d75c10 ffffff81 0117e028 ffffffc0
[   10.347931] 39c0  00d75c90 ffffff81 00000000 00000000 0acc3a00 ffffffc0 089c38e8 ffffffc0
[   10.347938] 39e0  00000000 00000000 0117e028 ffffffc0 00d75c10 ffffff81 00000000 00000000
[   10.347945] 3a00  0acc3a30 ffffffc0 089c0340 ffffffc0 00000000 00000000 0117e028 ffffffc0
[   10.347953] ---[ end trace 451d180e3f649ec2 ]---
[   10.350967] dw-hdmi-qp-cec dw-hdmi-qp-cec.4.auto: cec request irq thread failed
[   10.351027] dw-hdmi-qp-cec: probe of dw-hdmi-qp-cec.4.auto failed with error -22

unifreq, 2023-07-09:
The relevant driver is “drivers/gpu/drm/bridge/synopsys“，
if fallback code to commit 68767361ad510a19532962ed3c3f75bd0830d879
then the problem disappears
[image]

RadxaYuntian, 2023-07-10:
Current rkr4.1 branch does not support RK3588 products. Currently only ROCK 3C and CM3S are supported in this branch.

RadxaYuntian, 2023-07-10:
@RadxaStephen told me he is already using rkr4.1 branch in Yocto to support ROCK 5.