radxa/kernel #236: Issue with DSI (RPi 7" LCD) display touch interface

BigMuscle85, 2023-10-30:
Hello,
I'm trying to make the official Raspberry Pi display to work on Radxa Rock 4B+. The image over DSI wotk correctly, but touch interface is somehow buggy. Driver raspits_ft5426 is loaded but only many errors are logged. Touch does not work.

> raspits-ft5426: raspits_ft5426_probe: address = 0x38
> input: fts_ts as /devices/platform/ff110000.i2c/i2c-1/1-0038/input/input6
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: raspits_ft5426_work: checking touch ic, countdown: 8
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: raspits_ft5426_work: checking touch ic, countdown: 7
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: raspits_ft5426_work: checking touch ic, countdown: 6
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: raspits_ft5426_work: checking touch ic, countdown: 5
> raspits-ft5426: fts_check_fw_ver: Firmware version = 0.0.0
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: fts_read_td_status: get reg td_status failed, -6
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: fts_read_td_status: get reg td_status failed, -6
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: fts_read_td_status: get reg td_status failed, -6
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: fts_read_td_status: get reg td_status failed, -6
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: fts_read_td_status: get reg td_status failed, -6
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: fts_read_td_status: get reg td_status failed, -6
> raspits-ft5426: fts_i2c_read: i2c read error, -6
> raspits-ft5426: fts_read_td_status: get reg td_status failed, -6
> raspits-ft5426: fts_i2c_read: i2c read error, -6 

I only connected DSI cable, 5V and GND cables. The following is returned by i2cdetect:
```
rock@rock-pi-4b-plus:~$ sudo i2cdetect -l
i2c-1   i2c             rk3x-i2c                                I2C adapter
i2c-0   i2c             rk3x-i2c                                I2C adapter
i2c-9   i2c             DesignWare HDMI                         I2C adapter
rock@rock-pi-4b-plus:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
But when I connect also SDA/SCL pins to I2C7_SDA and I2C7_SCL pins (https://wiki.radxa.com/Rockpi4/hardware/gpio) then the touch starts working. Although the "read error -6" errors are still logged and it randomly sends touch event to right bottom corner.

The only enabled overlay is for Raspberry Pi display.
I'm trying to understand what's going on. Could there be some bug in the kernel? Why touch starts working when connected to I2C7 pins although the overlay is for I2C1?

RadxaYuntian, 2023-10-30:
Overlay was initially added in https://github.com/radxa/kernel/commit/0c2da8a6b711cbeb88c95a429f4330e8dc225d63

BigMuscle85, 2023-10-30:
Yes, this overlay is active.
I tried to control touch interface directly from userspace without the kernel module and it behaves the same. It works only if SDA/SCL pins are connected to some I2C pins on the board (although this pins are not activated via overlay). Otherwise returns errno 6 on write.

```
#include <errno.h>
#include <fcntl.h>
#include <stdio.h>
#include <unistd.h>

#include <linux/i2c-dev.h>
#include <sys/ioctl.h>

int main() {
	int fd = open("/dev/i2c-1", O_RDWR);
	if(fd < 0) {
		printf("open: Error %d\n", errno);
	} else {
		int ret = ioctl(fd, I2C_SLAVE, 0x38);
		if(ret < 0) {
			printf("ioctl: Error %d\n", errno);
		} else {
			while(true) {
				unsigned char reg_addr = 2; // status register					
				ret = write(fd, &reg_addr, 1);
				if(ret != 1) {
					printf("write error, ret = %d, err = %d\n", ret, errno);
				} else {				
					unsigned char res = 0;
					ret = read(fd, &res, 1);
					// res is 1 on touch
					printf("read ret = %d, res = %d, errno = %d\n", ret, res, errno);
				}
				
				usleep(17*1000);
			}
		}
		close(fd);
	}

	return 0;
}
```

BigMuscle85, 2023-10-31:
I am starting to feel that the problem is not in RPi display itself, but something wrong is with I2C1 bus. Currently, I started getting tons of the following errors and even "i2cdetect -y 1" was very slow. I did "sudo reboot" but it did not help and the same situation was repeating even after reboot. I had to disconnect the power completely.

> [   40.771872] rk3x-i2c ff110000.i2c: timeout, ipd: 0x00, state: 1
> [   40.771911] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011: -110
> [   40.875315] rk3x-i2c ff110000.i2c: timeout, ipd: 0x00, state: 1
> [   40.875386] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011: -110
> [   40.978649] rk3x-i2c ff110000.i2c: timeout, ipd: 0x00, state: 1
> [   40.978721] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011: -110
> [   41.082009] rk3x-i2c ff110000.i2c: timeout, ipd: 0x00, state: 1
> [   41.082079] es8316 1-0011: ASoC: error at soc_component_read_no_lock on es8316.1-0011: -110

BigMuscle85, 2023-11-01:
Another weird thing from the debugging. 
i2cdetect normally shows the following:

```
rock@rock-pi-4b-plus:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

But when I start communicating with the touch interface (0x38), strange things starts happening and i2cdetect shows random devices on every run:

```
rock@rock-pi-4b-plus:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- 66 67 -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
rock@rock-pi-4b-plus:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- 0e --
10: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
rock@rock-pi-4b-plus:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- 38 -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- 47 48 49 -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

It can be seen with simple `watch -n 0.5 "sudo i2cdetect -y 1"`
When I don't connect I2C6 SDA/SCL pins then address 38 is alternately visible and invisible on every other run which maybe the cause of errors.

I also tried different LCD display and it behaves in the same wrong way.