radxa/kernel #295: Missing CONFIG_ROCKCHIP_OPP in rk3308_linux_defconfig in branch linux-6.1-stan-rkr1

pavhofman, 2024-04-04:
Missing CONFIG_ROCKCHIP_OPP=y in rk3308_linux_defconfig results in no compilation of rockchip_opp_select.c which results in cpufreq features not available, operating points defined in rk3308.dtsi ignored, and all cores running only at the initial 800MHz.

RadxaYuntian, 2024-04-07:
We do not support RK3308 on linux-6.1-stan-rkr1 branch right now. Currently we are using [upstream kernel](https://github.com/radxa-repo/bsp/blob/main/linux/.stable-6.1/fork.conf#L1-L8) for RK3308 based devices.

closed 2024-04-07 by RadxaYuntian

pavhofman, 2024-04-07:
I am afraid the upstream is broken in I2S - DMA integration https://forum.radxa.com/t/rock-pi-s-radxa-kernel-linux-6-1-stan-rkr1-inactive-usb-host/20552/4

RadxaStephen, 2024-04-08:
> Missing CONFIG_ROCKCHIP_OPP=y in rk3308_linux_defconfig results in no compilation of rockchip_opp_select.c which results in cpufreq features not available, operating points defined in rk3308.dtsi ignored, and all cores running only at the initial 800MHz.

Thanks. I added it. https://github.com/radxa/kernel/commit/6d4edb4439e8f327975b2e09552c97369f9c8385