radxa/kernel #298: rk-6.1-rkr1: captured video from IMX415 at MIPI/CSI not connected to ISP block, on rk3588s platform

enok71, 2024-04-12:
On my rock 5a I can capture video from my IMX415 camera connected to the 4 lane CSI port if I capture data from /dev/video11 which is the raw video stream. But trying to get data from /dev/video[01] (mainpath/selfpath, after ISP) doesn't work. The reason seems to be related to the fact that device node /dev/v4l-subdev4 (rkcif-mipi-lvds2) is not created at boot. There seems to be no connection from rkcif to rkisp. All this worked in 5.10 but doesn't work in 6.1. I suspect it is an issue for all rk3588s platforms.

See also discussion and details [here](https://forum.radxa.com/t/dev-video-01)

Revision tested: [27f6561551948be6350c9bbd549c84acb2e559af](https://github.com/armbian/linux-rockchip/commit/27f6561551948be6350c9bbd549c84acb2e559af)

RadxaYuntian, 2024-04-15:
We have not officially released any Rockchip 6.1 SDK based image, so at this moment we won't take support any questions about it. We ourselves are still trying to figure this one out.

@Ken-Vamrs please double check if this is an issue on our system.

enok71, 2024-04-15:
I appreciate your efforts. There is a message at boot about dependency cycle. Maybe that is a clue. Let me know if I can help.

```
[  +0.024271] platform csi2-dphy0: Fixed dependency cycle(s) with /mipi2-csi2
[  +0.000166] platform mipi2-csi2: Fixed dependency cycle(s) with /rkcif-mipi-lvds2
[  +0.000135] platform rkcif-mipi-lvds2-sditf: Fixed dependency cycle(s) with /rkisp0-vir0
```

ivancheung2019, 2024-06-12:
@enok71 problem is fixed? I also got a problem on official image. the mipi video can show on cheese but is greenish and gain control is not running..

RadxaYuntian, 2024-06-12:
@ivancheung2019 this issue is about the upcoming 6.1 kernel. There is no official 6.1 image at the moment.

ivancheung2019, 2024-06-12:
> @ivancheung2019 this issue is about the upcoming 6.1 kernel. There is no official 6.1 image at the moment.

but I also facing such problem in B18 Debian KDE and ubuntu KDE pre-built image.. and test cheese.

RadxaYuntian, 2024-06-12:
It is still the wrong place to discuss this issue. You can create a [forum post](https://forum.radxa.com/) and tag `@radxa` in our forum to get technical support there.

Please also make sure you are using the Debian image (Ubuntu is unsupported) and performed [system upgrade](https://docs.radxa.com/en/template/sbc/os-config/rsetup#system-update) before you test the camera.