radxa/kernel #359: mtk-v5.15-dev-rity-kirkstone-v24.0 doesnt compile

Rippanda12, 2024-08-29:
```bash
drivers/soc/mediatek/hdmirx/hdmi_rx2_cmd.c:27:10: fatal error: mtk_hdmirx.h: No such file or directory
   27 | #include <mtk_hdmirx.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [scripts/Makefile.build:288: drivers/soc/mediatek/hdmirx/hdmi_rx2_cmd.o] Error 1
```

Adding this to [drivers/soc/mediatek/hdmirx/Makefile](https://github.com/radxa/kernel/blob/b09773ff99ba2a40b4abf89aae1ccd1702d96c72/drivers/soc/mediatek/hdmirx/Makefile#L6) makes it compile
```Makefile
ccflags-y += -I$(srctree)/drivers/soc/mediatek/hdmirx/
```

RadxaStephen, 2024-10-10:
Can you show me your full build command?

Rippanda12, 2024-10-12:
https://github.com/BredOS/sbc-pkgbuilds/blob/f9fd9a1eaf4e922b4516b052762835ca3d21c044/linux-mediatek/PKGBUILD#L30-L44

https://github.com/BredOS/sbc-pkgbuilds/blob/main/linux-mediatek/config