radxa/kernel #378: Which tag shoud I use for rock pi 4b

markhsumark, 2024-11-12:
Hi, I am a student researching TrustZone with a keen interest in integrating GPU functionality within TrustZone environments.

I’m looking for a Linux kernel for the Rock Pi 4B to replace the existing Linux kernel in the OP-TEE project. I would like to know which branch or tag would be most suitable and stable.

Thanks.

wildblue, 2025-01-23:
Would like to get the same answer for my rock-pi-4b-plus.

BigMuscle85, 2025-02-06:
If you use bsp to build the kernel for Rock Pi 4B+, it builds the branch [linux-5.10-gen-rkr3.4](https://github.com/radxa/kernel/tree/linux-5.10-gen-rkr3.4)
There is no other kernel for this board. Although version 6 might work, it lacks a lot of basic features on this board (e.g. DSI etc.)